spectral-cockpit/opusreader2 #23: Upon discussion: how to implement parallelized/chunked reading

philipp-baumann, 2022-08-25:
Currently, `read_opus_file()` reads a single file. It takes about 20 milliseconds to parse all relevant spectral blocks. This is not a lot, but still it would be nice to feature some feature to read in parallel.

The idea of this feature is that instead of distributing single file reads across cores or other parallelized instances, we chunk/split a file list according to the registered parallel instances.

To allow maximal flexibility, we should rely on the R "future" framework. As we don't want any other dependencies than base R, there seems two suitable options for us:

1. add a `parallel_chunk` (or other suitable name) argument to `read_opus()`, the wrapper. Then, check for the availability of {future.apply} using `require()`, which stops with a graceful message if not available, and parses in parallel if the package is available.
2. Create a separate package, e.g. {readopus2prod}, which contains advanced helper utilities, and has {opusreader2} as a dependency.

pierreroudier, 2022-09-07:
@philipp-baumann Just noting that my PR https://github.com/spectral-cockpit/opusreader2/pull/31 only implements the multiple file read sequentially, I have not added the option to add a `cluster` to `pbapply::pblapply` so several nodes can be used.

Like you, I have a feeling that the overheads would be way too high, and using chunks would be a better option. 

In my experience (as a user), a progress bar is nice when you start working with thousands of file, but of course it's not a *must* have.

philipp-baumann, 2022-09-07:
yes. agree :-) if I have time tomorrow I can draft something tomorrow and we can further discuss.

philipp-baumann, 2022-09-07:
progress bar is nice though. if one has to wait (as you say maybe 10000's of files, it's is nice to see how long one has to wait :)).

philipp-baumann, 2022-09-07:
if we use future, we let the user decide how to parallelize. The developer only decides what to parallelize.

philipp-baumann, 2022-09-08:
@ThomasKnecht , we have now `read_opus()` at top-level in the main branch. If we introduce let's say parallel chunking functionality via the `chunk_parallel` argument, we would need another top-level function. I see the following solution:

1. we rename `read_opus()` to `read_opus_impl()` 
2. we introduce a new `read_opus`, that dispatches to parallel if `chunk_parallel = TRUE`

Do you agree?

ThomasKnecht, 2022-09-12:
see #32

philipp-baumann, 2022-09-12:
> see #32

I like the `opus_apply()` style dispatch. Now what we need is a chunked version. Let's say we have something like this

```{R}
# register all available cores in multicore mode
library("future.apply")
plan(multiprocess) # or multicore if UNIX and a non-interactive environment
availableCores()
# then call `read_opus()` with parallel option, which then assigns to correct parallization mode
read_opus(dsn, parallel = TRUE)
```

ThomasKnecht, 2022-12-03:
a solution is implemented here: #47

ThomasKnecht, 2022-12-27:
can we close this?

closed 2023-02-01 by philipp-baumann