spectral-cockpit/opusreader2 #27: Introduce top-level `read_opus`

philipp-baumann, 2022-09-06:
#25

ThomasKnecht, 2022-09-09:
you are right, the output is kind of an issue. i think @philipp-baumann  thought about a filter option. but i see what you mean, we could add a param = T/F option to get the full output or just the data...
i like your idea of returning a matrix if multiple files are read :)

philipp-baumann, 2022-09-09:
Yes, I imagine a short iteration of the above:

Because `read_opus()` is high-level, we need a few short and concise arguments, that unleash the power in straight-forward manner, for simple and quite advanced use cases. In that sense it is worth to invest some time in design thinking.

- On individual measurements, there is so-called "in"-filters and "out_filters", specified with `filter_in` and `filter_out`. In @pierreroudier 's handy filter above would be in the `out_filter` class. There is many of these filters, but they will receive an extensive and simple description.

To grab specific elements, like `x$refl_no_atm_comp$data`, we have already thought of an out filter wrapper. 

What we also have to consider is a ``. 

Regarding chunking: I would go with the registered cores. Parallel resource management should be done by the user. We can further simplify the function interface by adding one main parallel argument, like `parallel_args`, that are individually dispatched.

philipp-baumann, 2022-09-09:
so a short proposal, although we probably need to update:

```{R}
res <- read_opus(
  # Pass a vector of DSN, can be files or raw sources, can be more than one
  dsn = my_dsn,
  filter_in = my_named_list_of_filters_in,
  filter_out = my_named_list_of_filters_out,
  datatype_out = my_list_of_datatypes_out,
  parallel_args = my_named_list_of_parallel_args,
  progress_bar = FALSE
)
```

philipp-baumann, 2022-09-09:
> you are right, the output is kind of an issue. i think @philipp-baumann thought about a filter option. but i see what you mean, we could add a param = T/F option to get the full output or just the data... i like your idea of returning a matrix if multiple files are read :)

Let's fit this `param` as early in the arg list for `filter_out`

philipp-baumann, 2022-09-10:
> filter_out

Another iteration, based on the discussion with @ThomasKnecht . I think the previous draft was a bit too complicated

```
res <- read_opus(
  dsn = my_dsn,
  output_datatype,
  output_path = NULL,
  parallel = FALSE,
  progress_bar = FALSE
)
```

ThomasKnecht, 2022-09-10:
instead of ouput_datatype -> data_only = F

philipp-baumann, 2022-09-10:
+1 for this Mr. Propper.

philipp-baumann, 2022-09-10:
```
res <- read_opus(
  dsn = my_dsn,
  data_only = FALSE,
  output_path = NULL,
  parallel = FALSE,
  progress_bar = FALSE
)
```

pierreroudier, 2022-09-12:
It;s unclear to me what `output_path` means

philipp-baumann, 2022-09-12:
> It;s unclear to me what `output_path` means

 @param output_path optional storage path for the parsed output. Default is
#' `NULL`, which only returns the parsed data as an in-memory R object.

philipp-baumann, 2022-09-12:
@pierreroudier do you have a better name suggestion than `output_path`?

ThomasKnecht, 2022-09-12:
I implemented the data_only as well as the parallel parts

@philipp-baumann @pierreroudier what do you think of that?

the data_only I also implemented in the read_opus_impl -> I think this is a nice to have in the base functionality anyways...

I am not so happy about my implementation of the data_only. it seems rather complicated... maybe you have a nicer suggestion?

ThomasKnecht, 2022-09-12:
I also fixed a bug in the prep_spectra function ;)

philipp-baumann, 2022-09-12:
> I implemented the data_only as well as the parallel parts
>
> @philipp-baumann @pierreroudier what do you think of that?
>
> the data_only I also implemented in the read_opus_impl -> I think this is a [...]

Awesome, thanks for this! short comments:
- we will need to do chunked, reading, so I will add some chunking logic to it (when 1 file takes about 20ms to read, the overhead is other wise too high that this has any realistic benefit
- I will think about `data_only` and come to you later this week.

pierreroudier, 2022-09-12:
Sorry, I am still a bit confused by `output_path`, is that an option to write the data to disk? In which case, what format would be used to write to file(s)?

I'd be tempted to leave such a functionality outside `read_opus`, but maybe I'm missing something (large collections that don;t fit into memory?).

For `data_only`, what is the type returned? Matrix, or still a (trimmed) list?

pierreroudier, 2022-09-12:
Opinionated comment on the arg names -- I like to keep them short, eg `progress` rather than `progress_bar`, maybe `folder` rather than `output_path`, etc.

philipp-baumann, 2022-09-13:
> Sorry, I am still a bit confused by `output_path`, is that an option to write the data to disk? In which case, what format would be used to write to file(s)?
>
> I'd be tempted to leave such a [...]

Maybe it was a bit many things in one issue, or respectively the merge that followed ;-) Yes indeed, `output_path` is an option to write data to disk. This could be for example in `.Rds` format, or in `.json`; the latter needs another transformation module. Another option is having a zarr-backend, which would mean a chunked n-dimensional array that represents all spectral data. I would like to have such a feature. If we stream data somewhere and have some quality control pipeline afterwards, we can at some point work in cloud environments such as s3 storages etc. At the moment I am not thinking about big data with such simple spectra, but I have sometimes project directories with few thousand OPUS files to be parsed.

pierreroudier, 2022-09-13:
> Maybe it was a bit many things in one issue, or respectively the merge that followed ;-) Yes indeed, `output_path` is an option to write data to disk. This could be for example in `.Rds` format, or [...]

Ok, got it. I think it would be best to develop the feature as a dedicated function, then once it's properly designed, consider its integration in `read_opus`. It's an advanced feature. 

> For `data_only`, at the moment it is still a list. We have to think about some good logic. @pierreroudier @ThomasKnecht shall we imply with `data_only` a matrix, which might be a good simplification in one argument?

My argument for this one is that users will either want to have the full metadata (in which case a list shall be returned), or want to quickly assemble data for modelling, in which case a matrix shall be returned.

For the latter, [there is code pretty much ready to go in the old `{opusreader}` package](https://github.com/pierreroudier/opusreader/blob/15b47d1f6887d37d043c193711a23dd838f9e611/R/opus_read_file.R#L104-L170).

philipp-baumann, 2022-09-14:
I generally like the matrix-support. To keep track of features and the development, I will move on with opening separate issue tickets. Like this it will be more simple to keep track and have a clean git workflow without too much wizardry.

ThomasKnecht, 2022-12-03:
we removed the output_path. we want to make a separate function for writing the data into json files or whatever other format is possible

philipp-baumann, 2022-12-07:
@pierreroudier , as @ThomasKnecht indicated above, the design decision we are making is to keep the interface of  `read_opus()` as generic as possible and with lowest possible complexity in arguments. This will enable us to have a rock solid interface that does not change over time, that serves its purpose for a broad community of users and contributors. Over time, there will be for sure options to customize specific wishes in helper functions, that are easier to maintain and are more specific to a particular spectroscopy workflow of individual organizations and companies.

philipp-baumann, 2022-12-07:
Such extended functionalities are for example #36 , #44 .
Closing because `read_opus()` and functions at various levels have been split by a new modular design with #47

philipp-baumann, 2022-12-07:
#47

closed 2022-12-07 by philipp-baumann