spectral-cockpit/opusreader2 #67: Tr matrix with relative times in OPUS exported MAT file

luukvandervelden, 2023-01-28:
I'm working with the python codebase of opusreader and am trying to figure out the time data of my spectra.
I have a two channels of timeseries of absorbance spectra and am able to extract them from the .0 files correctly.
When I export the data from OPUS to a MAT file, both the absorbance data AB matrix and the time data Tr matrix are exported. 
Now I wonder how to get this matrix with relative time data from the binary files. Considering this R repository is rather active, I hope someone can help me. Does anyone have experience with this?

Some info about the Tr matrix. It contains 5 columns. I figured out that column 2 and 3 belong to the absorbance spectra (dual channel, a simultaneous ATR and transmissions measurement). The first 10 rows look like this:

`0	0.0529999732971191
1.06299996376038	1.11600017547607
2.12600016593933	2.17900013923645
3.19000005722046	3.24300003051758
4.25300025939941	4.30599975585938
67.9700012207031	68.0230026245117
68.0670013427734	68.1200027465820
68.1630020141602	68.2160034179688
68.2600021362305	68.3130035400391
68.3570022583008	68.4100036621094`

the first 5 datapoints are just after depositing the sample. Then there is a wait time of approximately 60 seconds, after which the measurement starts with a reasonably high acquisition rate.

philipp-baumann, 2023-01-29:
Hi @luukvandervelden 

Thanks for opening, and welcome here!
I think you are decribing a nice feature.
Could you share an example OPUS binary here?

Cheers,
Philipp

luukvandervelden, 2023-02-02:
Hi @philipp-baumann 

I've included a recording of ethanol in the attached zip. I took an effort to scrap the name of the operator (and verified in Opus and using BrukerOpusReader that the file is not corrupted)

[PyOReader-Testsample-EtOH.zip](https://github.com/spectral-cockpit/opusreader2/files/10569855/PyOReader-Testsample-EtOH.zip)

This plot shows the spectra of the two channels, with the time on the z-axis. Not a lot happens over time, except for some evaporation of the ethanol, but it is just as a simple example:

[image]

There is a total of 405 spectra per channel, so there must be 405 timesteps somewere as well. We saw in OPUS the TRC block (?): 

[image]

If we open it we see a matrix with the times. The SRT column contains the relative times for the ATR channel, the ERT for the Transmission channel:
[image]

I hope this example is clear. Thank you for your time!

Cheers, Luuk

philipp-baumann, 2023-02-10:
Hi Luuk, thanks for the excellent example. I finally have some free time to work a bit on it and get some stuff testing. I'll get back to you once I make some progress/find something useful to share here and/or implement.

philipp-baumann, 2023-02-10:
Cool graph btw :100:

philipp-baumann, 2023-02-11:
Just briefly reporting current state without time support.

``` r
devtools::load_all()
#> ℹ Loading opusreader2

file <- system.file("extdata/new_data", "PyOReader-Testsample-EtOH.0",
  package = "opusreader2"
)

data <- read_opus(dsn = file)
#> Error: Unknown header entry.
#>  The following 'composite key' is not yet mapped in the {opusreader2} key-value map of the header:
#>  * "b16-c28-t80-a0" 
#> We encourage your contribution to feature this new OPUS block by opening a new issue on
#>      https://github.com/spectral-cockpit/opusreader2/issues 
#> Please
#>  1. report reproducibly, using short code with {opusreader2}
#>       (recommended: https://reprex.tidyverse.org) 
#>  2. describe briefly
#>    a) Bruker instrument used
#>    b) equipment
#>    c) measurement mode and spectral blocks saved (OPUS settings)
#>    d) OPUS software version
#>    e) your general workflow for spectroscopic diagnostics
#>  3. provide an example OPUS binary file uploaded for public access
#>     on GitHub (best in issue)
#>  4. to facilitate widespread support of Bruker devices in open source
#>     based infrastructure, show the data blocks as print screens in the
#>     Bruker OPUS software (right-click in Viewer). Please upload the
#>     contents of all OPUS blocks in individual screenshots.
```

<sup>Created on 2023-02-11 by the [reprex package](https://reprex.tidyverse.org) (v2.0.1)</sup>

philipp-baumann, 2023-02-11:
wip in https://github.com/spectral-cockpit/opusreader2/tree/feat-67-time-resolved-spectra

philipp-baumann, 2023-02-13:
for the moment, I need to name the `"unknown_..."` block names.

``` r
devtools::load_all()
#> ℹ Loading opusreader2

file <- system.file("extdata/new_data", "PyOReader-Testsample-EtOH.0",
  package = "opusreader2"
)

data <- read_opus(dsn = file)
#> Warning in readBin(con, what = "character", n = n, size = 1, endian =
#> "little"): incomplete string at end of file has been discarded

#> Warning in readBin(con, what = "character", n = n, size = 1, endian =
#> "little"): incomplete string at end of file has been discarded

names(data)
#>  [1] "sc_ref_data_param"          "sc_ref"                    
#>  [3] "ig_ref_data_param"          "ig_ref"                    
#>  [5] "optics"                     "optics_ref"                
#>  [7] "acquisition_ref"            "fourier_transformation_ref"
#>  [9] "fourier_transformation"     "sample"                    
#> [11] "acquisition"                "instrument_ref"            
#> [13] "instrument"                 "history"

str(data$sc_ref)
#> List of 10
#>  $ block_type     : int 11
#>  $ channel_type   : int 4
#>  $ text_type      : int 0
#>  $ additional_type: int 0
#>  $ offset         : int 2803976
#>  $ next_offset    : int 2810640
#>  $ chunk_size     : int 1666
#>  $ block_type_name: chr "sc_ref"
#>  $ data           : num [1, 1:1665] 0.225 0.212 0.207 0.221 0.235 ...
#>   ..- attr(*, "dimnames")=List of 2
#>   .. ..$ : NULL
#>   .. ..$ : chr [1:1665] "3805.05761897564" "3803.12905454636" "3801.20049011707" "3799.27192568779" ...
#>  $ wavenumbers    : num [1:1665] 3805 3803 3801 3799 3797 ...
#>  - attr(*, "class")= chr "data"
str(data$sc_ref_data_param)
#> List of 9
#>  $ block_type     : int 27
#>  $ channel_type   : int 4
#>  $ text_type      : int 0
#>  $ additional_type: int 0
#>  $ offset         : int 2803800
#>  $ next_offset    : int 2803976
#>  $ chunk_size     : int 176
#>  $ block_type_name: chr "sc_ref_data_param"
#>  $ parameters     :List of 10
#>   ..$ DPF:List of 4
#>   .. ..$ parameter_name     : chr "DPF"
#>   .. ..$ parameter_name_long: chr "Data Point Format"
#>   .. ..$ parameter_value    : int 1
#>   .. ..$ parameter_type     : chr "int"
#>   ..$ NPT:List of 4
#>   .. ..$ parameter_name     : chr "NPT"
#>   .. ..$ parameter_name_long: chr "Number of Data Points"
#>   .. ..$ parameter_value    : int 1665
#>   .. ..$ parameter_type     : chr "int"
#>   ..$ FXV:List of 4
#>   .. ..$ parameter_name     : chr "FXV"
#>   .. ..$ parameter_name_long: chr "Frequency of First Point"
#>   .. ..$ parameter_value    : num 3805
#>   .. ..$ parameter_type     : chr "float"
#>   ..$ LXV:List of 4
#>   .. ..$ parameter_name     : chr "LXV"
[... 96 lines cut ...]
#>     shiny         1.7.4   2022-12-15 [1] RSPM (R 4.1.0)
#>     stringi       1.7.12  2023-01-11 [1] CRAN (R 4.1.3)
#>     stringr       1.5.0   2022-12-02 [1] CRAN (R 4.1.3)
#>     styler        1.5.1   2021-07-13 [1] CRAN (R 4.1.1)
#>     testthat    * 3.1.6   2022-12-09 [1] CRAN (R 4.1.3)
#>     tibble        3.1.8   2022-07-22 [1] CRAN (R 4.1.3)
#>     urlchecker    1.0.1   2021-11-30 [1] CRAN (R 4.1.3)
#>     usethis       2.1.6   2022-05-25 [1] CRAN (R 4.1.3)
#>     utf8          1.2.3   2023-01-31 [1] CRAN (R 4.1.3)
#>     vctrs         0.5.2   2023-01-23 [1] CRAN (R 4.1.3)
#>     withr         2.5.0   2022-03-03 [1] CRAN (R 4.1.3)
#>     xfun          0.37    2023-01-31 [1] CRAN (R 4.1.3)
#>     xtable        1.8-4   2019-04-21 [1] CRAN (R 4.1.2)
#>     yaml          2.3.7   2023-01-23 [1] CRAN (R 4.1.3)
#> 
#>  [1] /home/philipp/R/x86_64-pc-linux-gnu-library/4.1
#>  [2] /opt/R/4.1.3/lib/R/library
#> 
#>  V ── Loaded and on-disk version mismatch.
#>  P ── Loaded and on-disk path mismatch.
#> 
#> ──────────────────────────────────────────────────────────────────────────────
```

<sup>Created on 2023-02-13 by the [reprex package](https://reprex.tidyverse.org) (v2.0.1)</sup>

philipp-baumann, 2023-02-13:
This is an example of a new key: https://github.com/spectral-cockpit/opusreader2/blob/fb738b383a48c2670035ba5e29a81e8c44f91c20/R/create_dataset.R#L36

philipp-baumann, 2023-02-21:
Hi @luukvandervelden found some stuff! are you familiar with R?

luukvandervelden, 2023-02-21:
Hi @philipp-baumann I am not familiar with R (I use Python and MATLAB), I am curious about what you've found and how you have found it!

philipp-baumann, 2023-02-21:
shall we have a short meeting on Friday or you prefer here?

luukvandervelden, 2023-02-21:
A meeting on Friday would be nice! My schedule this Friday is still completely empty, what time suits you?

philipp-baumann, 2023-02-21:
> shall we have a short meeting on Friday or you prefer here?

or let me prepare sth for Friday to put here # (have some imp stuff to do for opengis.ch )

luukvandervelden, 2023-02-21:
Sure, no rush, just give me a ping when you have some time @philipp-baumann

joshduran, 2024-05-14:
The data you are looking for is located in a mini-header present in all 3D data blocks.  The data is saved as a fixed-data structure that includes SRT (start time) and ERT (end time) for each 3D data block (among other parameters in array format).  You can see the full decoded data structure in my code:

https://github.com/joshduran/brukeropus

The relevant information is in `file/constants.py` and `file/file.py`

@luukvandervelden, since you are using python you can try my code out.  You access this data in the following way:

```python
from brukeropus import read_opus
data = read_opus(filepath)
# Absorbance data
x = data.a.x  # Wavenumbers
z = data.a.srt # Start time
y = data.a.y # Absorbance values (2D matrix)
```

Since the 3D header block does not include the three char parameter keys (e.g. SRT, ERT), I had to guess what they were and made a mistake for ERT.  To access ERT in the current version, use `data.a.tim`.  I will fix it in a future version to use `data.a.ert` to maintain consistency with OPUS.

[image]

luukvandervelden, 2024-06-27:
> @luukvandervelden, since you are using python you can try my code out. You access this data in the following way:

Hi @joshduran thank you for the ping and for the excellent news. I've tried your code and ittlooks like you have solved the puzzle!