spectral-cockpit/opusreader2 #81: Cannot open OPUS normalized spectra

Debora6991, 2023-03-20:
Hi!
I obtain this error when I try to read my OPUS files both with the single and multiple read approach,

[image]

[image]
I am using an old TENSOR 27 FTIR spectrometer, equipped with HTS-XT accessory  (OPUS 6.5)
What should I do?

philipp-baumann, 2023-03-20:
Hi @Debora6991 , could you upload your example OPUS file, e.g. "A1.1.0"

philipp-baumann, 2023-03-20:
Hi Debora, which email did you send it to?

Debora6991, 2023-03-20:
I answered directly by email because here the format file is not allowed. this is the e-mail address that appears 
AMCFBHYD6V27XJ5QHK5ND2WCEVWOREVBNHHGCQG6QI@reply.github.com.

Is this right? where should I attach the  .0 OPUS file?

philipp-baumann, 2023-03-20:
gosh stupid github, it used to support random binary files ;-) can you please send an email to spectral-cockpit at protonmail dot com ? thanks

Debora6991, 2023-03-20:
e-mail sent. 
Thankyou!

philipp-baumann, 2023-03-20:
got it! I'll give it a try this week. Could you please add screenshots of the OPUS software view (as described in informative message?) thanks

Debora6991, 2023-03-20:
Can you please add some extra info of what I should screenshot in OPUS
(6.5)?

philipp-baumann, 2023-03-20:
would be nice to have all spectral blocks screenshotted (right click on file...) :-) => here is a brief example: https://github.com/spectral-cockpit/opusreader2/issues/58

Debora6991, 2023-03-20:
> I want to just make sure everything we parse is double-checked on the OPUS
> side.

I screenshotted all spectral blocks, here they are

[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]

[image: image.png]

[image: image.png]

[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]

philipp-baumann, 2023-03-21:
Hi Debora, I do not see any images in your response above. Can you send them to me via mail?

philipp-baumann, 2023-03-21:
alternatively, can you give me a brief info here what all the block names are? I need an extra block type if you want your setting to be supported.

philipp-baumann, 2023-03-23:
@Debora6991 I think you did an email, but I did not receive, because it went to some GitHub address. If you want, I can post screenshots the file settings here?

Debora6991, 2023-03-23:
I sent the mail again (spectral-cockpit protonmail....), did it work?

philipp-baumann, 2023-03-23:
> I sent the mail again (spectral-cockpit protonmail....), did it work?
> […](#)

awesome, it got it. Will keep you posted.

philipp-baumann, 2023-03-24:
@Debora6991 from the screenshot I see you did vector normalization. I think I have seen this kind of mathematical black-box preprocessing used by Bruker before. ;-) Is it a vector normalization with L1 norm right? I think it is also multiplied with -1.

Debora6991, 2023-03-25:
Yes, exactly.

philipp-baumann, 2023-04-10:
just a short info to let you know. I am currently underway with my bike travelling. Eventually I have time I might push the necessary changes, but it can also be June until I am able to do so. Cheers

philipp-baumann, 2023-11-12:
Hi @Debora6991 finally there is a fix for your issue :-) 
With version 0.6.1 (see ) it should now work (see here https://github.com/spectral-cockpit/opusreader2/blob/main/NEWS.md )

``` r
library("opusreader2")
file <- "inst/extdata/new_data/issue81_A1.1.0"
data <- read_opus(dsn = file)
names(data[[1]])
#>  [1] "basic_metadata"            "ab_no_atm_comp_data_param"
#>  [3] "ab_no_atm_comp"            "sc_sample_data_param"     
#>  [5] "sc_sample"                 "sc_ref_data_param"        
#>  [7] "sc_ref"                    "optics"                   
#>  [9] "optics_ref"                "fourier_transformation"   
#> [11] "sample"                    "acquisition"              
#> [13] "instrument_ref"            "instrument"               
#> [15] "lab_and_process_param_raw" "info_block"               
#> [17] "quality_test_report"       "history"
data[[1]]$quality_test_report
#> $block_type
#> [1] 15
#> 
#> $channel_type
#> [1] 16
#> 
#> $text_type
#> [1] 96
#> 
#> $additional_type
#> [1] 0
#> 
#> $offset
#> [1] 28440
#> 
#> $next_offset
#> [1] 31200
#> 
#> $chunk_size
#> [1] 2760
#> 
#> $block_type_name
#> [1] "quality_test_report"
#> 
#> $parameters
#> $parameters$BID
#> $parameters$BID$parameter_name
#> [1] "BID"
#> 
#> $parameters$BID$parameter_name_long
#> NULL
#> 
#> $parameters$BID$parameter_value
#> [1] 6295567
#> 
#> $parameters$BID$parameter_type
#> [1] "int"
#> 
#> 
#> $parameters$NRP
#> $parameters$NRP$parameter_name
#> [1] "NRP"
#> 
#> $parameters$NRP$parameter_name_long
#> NULL
#> 
#> $parameters$NRP$parameter_value
#> [1] 1
#> 
#> $parameters$NRP$parameter_type
#> [1] "int"
#> 
#> 
#> $parameters$SIZ
#> $parameters$SIZ$parameter_name
#> [1] "SIZ"
#> 
#> $parameters$SIZ$parameter_name_long
#> NULL
#> 
#> $parameters$SIZ$parameter_value
#> [1] 124
#> 
#> $parameters$SIZ$parameter_type
#> [1] "int"
#> 
#> 
#> $parameters$TIT
#> $parameters$TIT$parameter_name
#> [1] "TIT"
#> 
#> $parameters$TIT$parameter_name_long
#> NULL
#> 
#> $parameters$TIT$parameter_value
#> [1] "Quality Test Report"
#> 
#> $parameters$TIT$parameter_type
#> [1] "str"
#> 
#> 
#> $parameters$E00
#> $parameters$E00$parameter_name
#> [1] "E00"
#> 
#> $parameters$E00$parameter_name_long
#> NULL
#> 
#> $parameters$E00$parameter_value
#> [1] "Quality Test Report"
#> 
#> $parameters$E00$parameter_type
#> [1] "str"
#> 
#> 
#> $parameters$F00
#> $parameters$F00$parameter_name
#> [1] "F00"
#> 
#> $parameters$F00$parameter_name_long
#> NULL
#> 
#> $parameters$F00$parameter_value
#> [1] 2012
#> 
#> $parameters$F00$parameter_type
#> [1] "int"
#> 
#> 
#> $parameters$Z00
#> $parameters$Z00$parameter_name
#> [1] "Z00"
#> 
#> $parameters$Z00$parameter_name_long
#> NULL
#> 
#> $parameters$Z00$parameter_value
#> [1] 748
#> 
#> $parameters$Z00$parameter_type
#> [1] "int"
#> 
#> 
#> 
#> attr(,"class")
#> [1] "parameter"
```

<sup>Created on 2023-11-12 with [reprex v2.0.2](https://reprex.tidyverse.org)</sup>

philipp-baumann, 2023-11-12:
It was a new parameter block, calling `"quality_test_report"`, which probably exists due to a OPUS macro used.

closed 2024-08-14 by philipp-baumann