spectral-cockpit/opusreader2 #82: Unknown header entry

Mahounakpon, 2023-04-28:
Firsly, i thank you for this package. 
I try to open opus data with opusreader2, but i received an error message like:

Error: Unknown header entry.
 The following 'composite key' is not yet mapped in the {opusreader2} key-value map of the header:
 * "b0-c0-t144-a1" 
 
I don't know how fixing this error.
 Thanks for yours answers.

philipp-baumann, 2023-04-29:
Hi, it looks like there is a data type in the OPUS file we have not yet specified. Could you share an example binary OPUS file here, so that I can have a look?

philipp-baumann, 2023-05-07:
@Mahounakpon just wondering whether your issue is still there. Can you share an OPUS file here?

Mahounakpon, 2023-05-09:
@philipp-baumann, hello! Here is an example of opus data with  data block (b0-c0-t144-a1...) . I appologize myself for this lateness. 
Thanks!
[Opus_test_unknown_entry.zip](https://github.com/spectral-cockpit/opusreader2/files/11428618/Opus_test_unknown_entry.zip)

Mahounakpon, 2023-05-22:
Okay!
Thanks

philipp-baumann, 2023-06-15:
Hi @DominiqueAhouanhode just short question back. What instrument are you using, what are you measuring and what data do you save?

aka 

```r
r$> file = system.file("extdata/test_data", "file-issue-82.0", package = "opusreader2")

r$> data = read_opus(file)
Error: Unknown header entry.
 The following 'composite key' is not yet mapped in the {opusreader2} key-value map of the header:
 * "b0-c0-t144-a1" 
We encourage your contribution to feature this new OPUS block by opening a new issue on
     https://github.com/spectral-cockpit/opusreader2/issues 
Please
 1. report reproducibly, using short code with {opusreader2}
      (recommended: https://reprex.tidyverse.org) 
 2. describe briefly
   a) Bruker instrument used
   b) equipment
   c) measurement mode and spectral blocks saved (OPUS settings)
   d) OPUS software version
   e) your general workflow for spectroscopic diagnostics
 3. provide an example OPUS binary file uploaded for public access
    on GitHub (best in issue)
 4. to facilitate widespread support of Bruker devices in open source
    based infrastructure, show the data blocks as print screens in the
    Bruker OPUS software (right-click in Viewer). Please upload the
    contents of all OPUS blocks in individual screenshots.
```

DominiqueAhouanhode, 2023-06-16:
Hi @philipp-baumann  we are using **BRUKER MPA**, 
We are measuring oisl and proteins composition,
**Absorbande data** are saved.
Thanks!

philipp-baumann, 2023-11-12:
Hi @DominiqueAhouanhode finally your reading issue should be resolved with the newest v0.6.0 release (see https://github.com/spectral-cockpit/opusreader2/blob/main/NEWS.md for details).
 
The composite key "b0-c0-t144-a1" belongs to a report block. Currently, we have classified it as `"report_unknown"` , not parsed, and returned as `NULL` element (see https://github.com/spectral-cockpit/opusreader2/blob/ed112befc014a4b4da44990dd10c1da41a89ca04/R/create_dataset.R#L29 ) . 

Here is a short confirmation.

``` r
library("opusreader2")
file <- "inst/extdata/new_data/issue82_Opus_test.0"
data <- read_opus(dsn = file)
names(data[[1]])
#>  [1] "basic_metadata"                  "ab_data_param"                  
#>  [3] "ab"                              "sc_sample_data_param"           
#>  [5] "sc_sample"                       "sc_ref_data_param"              
#>  [7] "sc_ref"                          "match_data_param"               
#>  [9] "match"                           "match_2_chn_data_param"         
#> [11] "match_2_chn"                     "optics"                         
#> [13] "optics_ref"                      "acquisition_ref"                
#> [15] "fourier_transformation_ref"      "fourier_transformation"         
#> [17] "sample"                          "acquisition"                    
#> [19] "instrument_ref"                  "instrument"                     
#> [21] "lab_and_process_param_processed" "info_block"                     
#> [23] "history"                         "report_unknown"
sessionInfo()
#> R version 4.1.3 (2022-03-10)
#> Platform: x86_64-pc-linux-gnu (64-bit)
#> Running under: Ubuntu 22.04.3 LTS
#> 
#> Matrix products: default
#> BLAS:   /usr/lib/x86_64-linux-gnu/blas/libblas.so.3.10.0
#> LAPACK: /usr/lib/x86_64-linux-gnu/lapack/liblapack.so.3.10.0
#> 
#> locale:
#>  [1] LC_CTYPE=en_US.UTF-8       LC_NUMERIC=C              
#>  [3] LC_TIME=de_CH.UTF-8        LC_COLLATE=en_US.UTF-8    
#>  [5] LC_MONETARY=de_CH.UTF-8    LC_MESSAGES=en_US.UTF-8   
#>  [7] LC_PAPER=de_CH.UTF-8       LC_NAME=C                 
#>  [9] LC_ADDRESS=C               LC_TELEPHONE=C            
#> [11] LC_MEASUREMENT=de_CH.UTF-8 LC_IDENTIFICATION=C       
#> 
#> attached base packages:
#> [1] stats     graphics  grDevices utils     datasets  methods   base     
#> 
#> other attached packages:
#> [1] opusreader2_0.6.0
#> 
#> loaded via a namespace (and not attached):
#>  [1] digest_0.6.33     withr_2.5.0       R.methodsS3_1.8.2 lifecycle_1.0.3  
#>  [5] magrittr_2.0.3    reprex_2.0.2      evaluate_0.22     rlang_1.1.1      
#>  [9] cli_3.6.1         rstudioapi_0.15.0 fs_1.6.3          R.utils_2.12.2   
#> [13] R.oo_1.25.0       vctrs_0.6.3       styler_1.10.2     rmarkdown_2.24   
#> [17] tools_4.1.3       R.cache_0.16.0    glue_1.6.2        purrr_1.0.2      
#> [21] xfun_0.40         yaml_2.3.7        fastmap_1.1.1     compiler_4.1.3   
#> [25] htmltools_0.5.6.1 knitr_1.43
```

<sup>Created on 2023-11-12 with [reprex v2.0.2](https://reprex.tidyverse.org)</sup>

I'll close once it works.

It would be great if you also upload screenshots of the OPUS software here, so that we can identify the new report blocks, try parsing it, and confirm that worked, too. As this block is a report block that is not so relevant because one wants to do analysis and modeling in R anyway, I suggest we can close once you confirm.

closed 2023-12-27 by philipp-baumann