spectral-cockpit/opusreader2 #92: Absorbance vs reflectance

davidbenncsiro, 2023-09-08:
`opusreader2` is looking good!

If I want to determine whether a sample represents absorbance or reflectance, should I look here?
~~~
opus2_result$acquisition$parameters$PLF
~~~

philipp-baumann, 2023-09-10:
> `opusreader2` is looking good!
> 
> If I want to determine whether a sample represents absorbance or reflectance, should I look here?
> 
> ```
> opus2_result$acquisition$parameters$PLF
> ```

Hey @davidbenncsiro , that is a good possibility. In case it is reflectance, you will have `"refl"` prefixes in `names(opus2_result)`.

davidbenncsiro, 2023-09-11:
Thanks @philipp-baumann. Other than the `refl` prefixes, would opus2_result$acquisition$parameters$PLF have something like `REFL` as a value in the case of sample reflectance?

philipp-baumann, 2023-09-11:
> Thanks @philipp-baumann. Other than the `refl` prefixes, would `opus2_result$acquisition$parameters$PLF` have something like `REFL` as a value in the case of sample reflectance?
> 

Here it is

```{r}
r$> library("opusreader2")
    # read a single file (one measurement)
    file <- opus_file()
    data_list <- read_opus(dsn = file)

r$> data_list[[1]]$acquisition$parameters$PLF
$parameter_name
[1] "PLF"

$parameter_name_long
[1] "Result Spectrum"

$parameter_value
[1] "RFL"

$parameter_type
[1] "str"

r$> sessioninfo::session_info()
─ Session info ───────────────────────────────────────────────────────────────────────────
 setting  value
 version  R version 4.1.3 (2022-03-10)
 os       Ubuntu 22.04.3 LTS
 system   x86_64, linux-gnu
 ui       X11
 language
 collate  en_US.UTF-8
 ctype    en_US.UTF-8
 tz       Europe/Zurich
 date     2023-09-11
 pandoc   2.9.2.1 @ /usr/bin/pandoc

─ Packages ───────────────────────────────────────────────────────────────────────────────
 package     * version    date (UTC) lib source
 cli           3.6.1      2023-03-23 [1] CRAN (R 4.1.3)
 opusreader2 * 0.5.0.9000 2023-09-11 [1] https://spectral-cockpit.r-universe.dev (R 4.1.3)
 sessioninfo   1.2.2      2021-12-06 [1] CRAN (R 4.1.2)

 [1] /home/philipp/R/x86_64-pc-linux-gnu-library/4.1
 [2] /opt/R/4.1.3/lib/R/library
```

> Also, since we're talking, are there plans for another tag/release in the near future? I'm currently using `master` which works well. I'm seeing some errors with from the official release. Can create another issue for this if it helps.

Yes, please make an issue :-)

philipp-baumann, 2023-09-11:
@davidbenncsiro regarding general progress. I am definitely interested in developing some new functionalities like a metadata helper. This is something I will for sure use in our soon-to-come business platform.

davidbenncsiro, 2023-09-11:
> @davidbenncsiro regarding general progress. I am definitely interested in developing some new functionalities like a metadata helper. This is something I will for sure use in our soon-to-come platform for spectroscopy end users. Patching I will for sure do asap, but for further developments I'd like to get some funding first.

Great to hear @philipp-baumann. I've started a conversation with CSIRO colleagues re: metadata of interest. One response is that what we get from the old `opusreader` metadata is OK, but there's an openness to additional metadata.

philipp-baumann, 2023-09-11:
> Great to hear @philipp-baumann. I've started a conversation with CSIRO colleagues re: metadata of interest. One response is that what we get from the old `opusreader` metadata is OK, but there's an [...]

sounds good! @ThomasKnecht and me have made a selection of relevant meta data already. Once this is ready we can also schedule a quick meeting to exchange. Cheers

closed 2023-09-11 by philipp-baumann