spectral-cockpit/opusreader2 #94: Error reading an Opus file

zecojls, 2023-11-06:
Hi, I'm trying to read an Opus file from a collaborator, and the following error is raised:
`"Error in ds_param[[1]] : subscript out of bounds"`

Reproducible code:
`test <- read_opus_single(dsn = "RT_01_1_23-02-21_13-23-54.0")`

The binary Opus file was sent by email to help with debugging. The data comes from a Bruker 2023 Alpha II.

I'm also attaching the screenshots of the file blocks.

Thanks

zecojls, 2023-11-06:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

philipp-baumann, 2023-11-07:
> Hi, I'm trying to read an Opus file from a collaborator, and the following error is raised: `"Error in ds_param[[1]] : subscript out of bounds"`
>
> Reproducible code: `test <- read_opus_single(dsn = [...]

Thanks @zecojls for the screenshots and the issue, @ThomasKnecht and me will investigate and hopefully fix the comming WE. Will keep you up to date. Kind regards, Philipp

philipp-baumann, 2023-11-12:
Hi @zecojls we were able to fix it :-) More infos about v0.6.0 that supports it are here https://github.com/spectral-cockpit/opusreader2/blob/main/NEWS.md

``` r
library("opusreader2")
file <- "inst/extdata/new_data/issue94_RT_01_1_23-02-21_13-23-54.0"
data <- read_opus(dsn = file)
names(data[[1]])
#>  [1] "basic_metadata"             "ab_no_atm_comp_data_param" 
#>  [3] "ab_no_atm_comp"             "sc_sample_data_param"      
#>  [5] "sc_sample"                  "sc_ref_data_param"         
#>  [7] "sc_ref"                     "optics"                    
#>  [9] "optics_ref"                 "acquisition_ref"           
#> [11] "fourier_transformation_ref" "fourier_transformation"    
#> [13] "sample"                     "acquisition"               
#> [15] "instrument_ref"             "instrument"                
#> [17] "history"
sessionInfo()
#> R version 4.1.3 (2022-03-10)
#> Platform: x86_64-pc-linux-gnu (64-bit)
#> Running under: Ubuntu 22.04.3 LTS
#> 
#> Matrix products: default
#> BLAS:   /usr/lib/x86_64-linux-gnu/blas/libblas.so.3.10.0
#> LAPACK: /usr/lib/x86_64-linux-gnu/lapack/liblapack.so.3.10.0
#> 
#> locale:
#>  [1] LC_CTYPE=en_US.UTF-8       LC_NUMERIC=C              
#>  [3] LC_TIME=de_CH.UTF-8        LC_COLLATE=en_US.UTF-8    
#>  [5] LC_MONETARY=de_CH.UTF-8    LC_MESSAGES=en_US.UTF-8   
#>  [7] LC_PAPER=de_CH.UTF-8       LC_NAME=C                 
#>  [9] LC_ADDRESS=C               LC_TELEPHONE=C            
#> [11] LC_MEASUREMENT=de_CH.UTF-8 LC_IDENTIFICATION=C       
#> 
#> attached base packages:
#> [1] stats     graphics  grDevices utils     datasets  methods   base     
#> 
#> other attached packages:
#> [1] opusreader2_0.6.0
#> 
#> loaded via a namespace (and not attached):
#>  [1] digest_0.6.33     withr_2.5.0       R.methodsS3_1.8.2 lifecycle_1.0.3  
#>  [5] magrittr_2.0.3    reprex_2.0.2      evaluate_0.22     rlang_1.1.1      
#>  [9] cli_3.6.1         rstudioapi_0.15.0 fs_1.6.3          R.utils_2.12.2   
#> [13] R.oo_1.25.0       vctrs_0.6.3       styler_1.10.2     rmarkdown_2.24   
#> [17] tools_4.1.3       R.cache_0.16.0    glue_1.6.2        purrr_1.0.2      
#> [21] xfun_0.40         yaml_2.3.7        fastmap_1.1.1     compiler_4.1.3   
#> [25] htmltools_0.5.6.1 knitr_1.43
```

<sup>Created on 2023-11-12 with [reprex v2.0.2](https://reprex.tidyverse.org)</sup>

You can just do an `remotes::install_github("spectral-cockpit/opusreader2` to get the newest version.
Once you confirm we can close the issue here.

zecojls, 2023-11-13:
Great! Thank you guys. Just to confirm with a quick silly question, what is the equivalent R list name to the Opus AB block?

zecojls, 2023-11-13:
Perfect, thank you. I think in the previous versions the description was not that clear but I'm glad to see you guys improved it a lot. I've been using the reflectance data and transforming it to apparent absorbance with log10(1/R) rather than natural log. Not sure if makes sense to properly document it as log10 for those not familiar with these transformations...

philipp-baumann, 2023-11-13:
just fixed with  https://github.com/spectral-cockpit/opusreader2/commit/850fa7357c780dfc4a7423431ee9189b6993d17f

closed 2023-11-13 by philipp-baumann