spectral-cockpit/opusreader2 #95: Support new header entry for OPUS file posted here: https://github.com/wincowgerDEV/OpenSpecy-package/issues/118

philipp-baumann, 2023-11-07:
refers to https://github.com/wincowgerDEV/OpenSpecy-package/issues/118

philipp-baumann, 2023-11-07:
``` r
devtools::load_all()
#> ℹ Loading opusreader2

file <- system.file("extdata", "new_data", "OpenSpecy-package_issue118.0",
  package = "opusreader2"
)

data <- read_opus(dsn = file)
#> Error: Unknown header entry.
#>  The following 'composite key' is not yet mapped in the {opusreader2} key-value map of the header:
#>  * "b15-c40-t0-a64" 
#> We encourage your contribution to feature this new OPUS block by opening a new issue on
#>      https://github.com/spectral-cockpit/opusreader2/issues 
#> Please
#>  1. report reproducibly, using short code with {opusreader2}
#>       (recommended: https://reprex.tidyverse.org) 
#>  2. describe briefly
#>    a) Bruker instrument used
#>    b) equipment
#>    c) measurement mode and spectral blocks saved (OPUS settings)
#>    d) OPUS software version
#>    e) your general workflow for spectroscopic diagnostics
#>  3. provide an example OPUS binary file uploaded for public access
#>     on GitHub (best in issue)
#>  4. to facilitate widespread support of Bruker devices in open source
#>     based infrastructure, show the data blocks as print screens in the
#>     Bruker OPUS software (right-click in Viewer). Please upload the
#>     contents of all OPUS blocks in individual screenshots.
```

<sup>Created on 2023-11-07 with [reprex v2.0.2](https://reprex.tidyverse.org)</sup>

philipp-baumann, 2023-11-12:
The single channels work so far, for the remaining blocks we need screenshots to try-error parsing settings.

{opusreader2} v0.6.0: see also https://github.com/spectral-cockpit/opusreader2/blob/main/NEWS.md

``` r
library("opusreader2")
file <- "inst/extdata/new_data/issue118_OpenSpecy-package.0"
data <- read_opus(dsn = file)
#> Warning: Unknown header entry.
#>  The following 'composite key' is not yet mapped in the {opusreader2} key-value map of the header:
#>  * "b15-c40-t0-a64" 
#> We encourage your contribution to feature this new OPUS block by opening a new issue on
#>      https://github.com/spectral-cockpit/opusreader2/issues 
#> Please
#>  1. report reproducibly, using short code with {opusreader2}
#>       (recommended: https://reprex.tidyverse.org) 
#>  2. describe briefly
#>    a) Bruker instrument used
#>    b) equipment
#>    c) measurement mode and spectral blocks saved (OPUS settings)
#>    d) OPUS software version
#>    e) your general workflow for spectroscopic diagnostics
#>  3. provide an example OPUS binary file uploaded for public access
#>     on GitHub (best in issue)
#>  4. to facilitate widespread support of Bruker devices in open source
#>     based infrastructure, show the data blocks as print screens in the
#>     Bruker OPUS software (right-click in Viewer). Please upload the
#>     contents of all OPUS blocks in individual screenshots.
#> Warning: Unknown header entry.
#>  The following 'composite key' is not yet mapped in the {opusreader2} key-value map of the header:
#>  * "b7-c132-t0-a64" 
#> We encourage your contribution to feature this new OPUS block by opening a new issue on
#>      https://github.com/spectral-cockpit/opusreader2/issues 
#> Please
#>  1. report reproducibly, using short code with {opusreader2}
#>       (recommended: https://reprex.tidyverse.org) 
#>  2. describe briefly
#>    a) Bruker instrument used
#>    b) equipment
#>    c) measurement mode and spectral blocks saved (OPUS settings)
#>    d) OPUS software version
#>    e) your general workflow for spectroscopic diagnostics
#>  3. provide an example OPUS binary file uploaded for public access
#>     on GitHub (best in issue)
#>  4. to facilitate widespread support of Bruker devices in open source
#>     based infrastructure, show the data blocks as print screens in the
#>     Bruker OPUS software (right-click in Viewer). Please upload the
#>     contents of all OPUS blocks in individual screenshots.
#> Warning: Unknown header entry.
#>  The following 'composite key' is not yet mapped in the {opusreader2} key-value map of the header:
#>  * "b23-c132-t0-a64" 
#> We encourage your contribution to feature this new OPUS block by opening a new issue on
#>      https://github.com/spectral-cockpit/opusreader2/issues 
#> Please
#>  1. report reproducibly, using short code with {opusreader2}
#>       (recommended: https://reprex.tidyverse.org) 
#>  2. describe briefly
#>    a) Bruker instrument used
#>    b) equipment
#>    c) measurement mode and spectral blocks saved (OPUS settings)
#>    d) OPUS software version
#>    e) your general workflow for spectroscopic diagnostics
#>  3. provide an example OPUS binary file uploaded for public access
#>     on GitHub (best in issue)
#>  4. to facilitate widespread support of Bruker devices in open source
#>     based infrastructure, show the data blocks as print screens in the
#>     Bruker OPUS software (right-click in Viewer). Please upload the
#>     contents of all OPUS blocks in individual screenshots.
#> Warning: Unknown header entry.
#>  The following 'composite key' is not yet mapped in the {opusreader2} key-value map of the header:
[... 252 lines cut ...]
#> locale:
#>  [1] LC_CTYPE=en_US.UTF-8       LC_NUMERIC=C              
#>  [3] LC_TIME=de_CH.UTF-8        LC_COLLATE=en_US.UTF-8    
#>  [5] LC_MONETARY=de_CH.UTF-8    LC_MESSAGES=en_US.UTF-8   
#>  [7] LC_PAPER=de_CH.UTF-8       LC_NAME=C                 
#>  [9] LC_ADDRESS=C               LC_TELEPHONE=C            
#> [11] LC_MEASUREMENT=de_CH.UTF-8 LC_IDENTIFICATION=C       
#> 
#> attached base packages:
#> [1] stats     graphics  grDevices utils     datasets  methods   base     
#> 
#> other attached packages:
#> [1] opusreader2_0.6.0
#> 
#> loaded via a namespace (and not attached):
#>  [1] digest_0.6.33     withr_2.5.0       R.methodsS3_1.8.2 lifecycle_1.0.3  
#>  [5] magrittr_2.0.3    reprex_2.0.2      evaluate_0.22     rlang_1.1.1      
#>  [9] cli_3.6.1         rstudioapi_0.15.0 fs_1.6.3          R.utils_2.12.2   
#> [13] R.oo_1.25.0       vctrs_0.6.3       styler_1.10.2     rmarkdown_2.24   
#> [17] tools_4.1.3       R.cache_0.16.0    glue_1.6.2        purrr_1.0.2      
#> [21] xfun_0.40         yaml_2.3.7        fastmap_1.1.1     compiler_4.1.3   
#> [25] htmltools_0.5.6.1 knitr_1.43
```

<sup>Created on 2023-11-12 with [reprex v2.0.2](https://reprex.tidyverse.org)</sup>

joshduran, 2024-05-11:
I am able to read this file from my python package: 

https://github.com/joshduran/brukeropus

Here is the file directory (header):
```
========================================================================================================================
                                     c:\Users\duranjm\Desktop\examples\files\test.0
                      Version: 920622.0    Directory start: 24    Max Blocks: 40    Num Blocks: 22

========================================================================================================================
                                                       Directory
Block Type                  Size (bytes)   Start (bytes)   Friendly Name
(0, 0, 0, 13, 0, 0)         480            24              Directory
(0, 0, 10, 0, 0, 0)         120            504             Sample Origin Parameters
(3, 3, 0, 10, 0, 0)         5808           624             Raman
(3, 2, 0, 1, 0, 0)          5808           6544            Reference Spectrum
(3, 2, 1, 1, 0, 0)          112            12352           Reference Data Status Parameters Spectrum
(3, 1, 0, 33, 0, 0)         3072           12464           Sample Unknown 3 33
(3, 1, 1, 33, 0, 0)         112            15536           Sample Data Status Parameters Unknown 3 33
(3, 1, 0, 65, 0, 0)         3072           15648           Sample Unknown 3 65
(3, 1, 1, 65, 0, 0)         112            18720           Sample Data Status Parameters Unknown 3 65
(3, 1, 0, 97, 0, 0)         3072           18832           Sample Unknown 3 97
(3, 1, 1, 97, 0, 0)         112            21904           Sample Data Status Parameters Unknown 3 97
(3, 2, 0, 33, 0, 0)         2960           22016           Reference Unknown 3 33
(3, 2, 1, 33, 0, 0)         112            24976           Reference Data Status Parameters Unknown 3 33
(3, 2, 0, 65, 0, 0)         2960           25088           Reference Unknown 3 65
(3, 2, 1, 65, 0, 0)         112            28048           Reference Data Status Parameters Unknown 3 65
(3, 2, 0, 97, 0, 0)         2960           28160           Reference Unknown 3 97
(3, 2, 1, 97, 0, 0)         112            31120           Reference Data Status Parameters Unknown 3 97
(0, 0, 2, 0, 0, 0)          136            31232           Instrument Status Parameters
(0, 0, 6, 0, 0, 0)          24             31368           Optical Parameters
(3, 3, 1, 10, 0, 0)         184            31392           Data Status Parameters Raman
(0, 0, 0, 0, 0, 1)          504            31576           Compound Information
(0, 0, 0, 0, 0, 5)          1696           32080           File Log
```

You can see there are several data blocks with unknown labels, but they parse as data arrays (or parameters where labeled).  I can look into trying to determine descriptive labels for these unknown type codes (33, 65, 97) so they can be presented to the user more cleanly.  If you aren't doing this already, you can see I break the block type integer into 6 parts (bit shifting).  This is the key to broadly classifying these blocks (i.e. reliably determine if parameter block, data block, 3D data block, etc...

here is the full output if it helps:

```
========================================================================================================================
                                     c:\Users\duranjm\Desktop\examples\files\test.0
                      Version: 920622.0    Directory start: 24    Max Blocks: 40    Num Blocks: 22

========================================================================================================================
                                                       Directory
Block Type                  Size (bytes)   Start (bytes)   Friendly Name
(0, 0, 0, 13, 0, 0)         480            24              Directory
(0, 0, 10, 0, 0, 0)         120            504             Sample Origin Parameters
(3, 3, 0, 10, 0, 0)         5808           624             Raman
(3, 2, 0, 1, 0, 0)          5808           6544            Reference Spectrum
(3, 2, 1, 1, 0, 0)          112            12352           Reference Data Status Parameters Spectrum
(3, 1, 0, 33, 0, 0)         3072           12464           Sample Unknown 3 33
(3, 1, 1, 33, 0, 0)         112            15536           Sample Data Status Parameters Unknown 3 33
(3, 1, 0, 65, 0, 0)         3072           15648           Sample Unknown 3 65
(3, 1, 1, 65, 0, 0)         112            18720           Sample Data Status Parameters Unknown 3 65
(3, 1, 0, 97, 0, 0)         3072           18832           Sample Unknown 3 97
(3, 1, 1, 97, 0, 0)         112            21904           Sample Data Status Parameters Unknown 3 97
(3, 2, 0, 33, 0, 0)         2960           22016           Reference Unknown 3 33
(3, 2, 1, 33, 0, 0)         112            24976           Reference Data Status Parameters Unknown 3 33
(3, 2, 0, 65, 0, 0)         2960           25088           Reference Unknown 3 65
(3, 2, 1, 65, 0, 0)         112            28048           Reference Data Status Parameters Unknown 3 65
(3, 2, 0, 97, 0, 0)         2960           28160           Reference Unknown 3 97
(3, 2, 1, 97, 0, 0)         112            31120           Reference Data Status Parameters Unknown 3 97
(0, 0, 2, 0, 0, 0)          136            31232           Instrument Status Parameters
(0, 0, 6, 0, 0, 0)          24             31368           Optical Parameters
(3, 3, 1, 10, 0, 0)         184            31392           Data Status Parameters Raman
[... cut ...]
T09='TEXT:Lot/Batch Number', T10='TEXT:Container Number', T11='TEXT:Tip', T12='TEXT:Threshold', T13='TEXT:Spectral
Ranges'});
Save File               2022/09/15 13:26:25 (GMT-4)
                        COMMAND_LINE Save (["::this::":Raman],
{DAP='C:\Users\Public\Documents\Bruker\OPUS_8.7.31\Data\Moffitt\LargePVfield, OEX='0',
SAN='4-01-01AJ_NIST3_0_0_20220915-110841_en.0', COF=130, INP='C:\Users\Public\Documents\Bruker\OPUS_8.7.31\METHODS',
IFP='C:\Users\Public\Documents\Bruker\OPUS_8.7.31\METHODS', INM='DEFAULT', IFN='DEFAULT'});
```