spectral-cockpit/opusreader2 #102: read_opus error

EstherDGoldstein-NOAA, 2023-12-20:
Hello,

Some R code that previously worked using the read_opus function no longer works. The files I am working with are from a Bruker MPA II FT-NIR multipurpose analyzer. 

The code that no longer works is:
data <- opusreader2::read_opus(dsn = file)
Where file is a specific file path. This particular test of the function was using just a single file path as a test to make it simple. 

The error that I receive is:
Error in UseMethod("calc_parameter_chunk_size") : 
  no applicable method for 'calc_parameter_chunk_size' applied to an object of class "parameter"

Any help is much appreciated.

Thank you!

philipp-baumann, 2023-12-23:
Hi Esther,

Can you please post your OPUS file here so that I can reproduce the error?

Cheers

philipp-baumann, 2023-12-23:
What is your `sessionInfo()` or `sessioninfo::session_info()` output?

philipp-baumann, 2023-12-23:
@EstherDGoldstein-NOAA do you remember which version worked? wonder if it really the same as #101 which would be a new bug we might have accidentally introduced. Would be good if you can help us diagnose by posting an OPUS file. Thank you :-)

philipp-baumann, 2023-12-23:
can confirm your issue, we have test running on files. I think I have messed up something. We had `devtools::check()` on GitHub actions before, but I updated the CI and apparently forgot to add. 

Here is my local output on the test files to confirm.

```r
> library(opusreader2)
> devtools::test()
ℹ Testing opusreader2
✔ | F W  S  OK | Context
✔ |          6 | create_dataset                                                                                
✔ |          4 | parse_header                                                                                  
✖ | 1        0 | read_opus                                                                                     
───────────────────────────────────────────────────────────────────────────────────────────────────────────────
Error (test-read_opus.R:2:3): all test files are parsed without warning
Error in `UseMethod("calc_parameter_chunk_size")`: no applicable method for 'calc_parameter_chunk_size' applied to an object of class "parameter"
Backtrace:
     ▆
  1. ├─testthat::expect_no_warning(read_opus(dsn = "../../inst/extdata/test_data")) at test-read_opus.R:2:3
  2. │ └─testthat:::expect_no_(...)
  3. │   └─testthat:::quasi_capture(enquo(object), NULL, capture)
  4. │     ├─testthat (local) .capture(...)
  5. │     │ └─rlang::try_fetch(...)
  6. │     │   └─base::withCallingHandlers(...)
  7. │     └─rlang::eval_bare(quo_get_expr(.quo), quo_get_env(.quo))
  8. └─opusreader2::read_opus(dsn = "../../inst/extdata/test_data")
  9.   └─opusreader2:::opus_lapply(dsn, data_only) at opusreader2/R/read_opus.R:152:5
 10.     └─base::lapply(...) at opusreader2/R/read_opus.R:237:3
 11.       └─opusreader2 (local) FUN(X[[i]], ...)
 12.         └─opusreader2::read_opus_single(x, data_only) at opusreader2/R/read_opus.R:240:7
 13.           └─opusreader2:::parse_opus(raw, data_only) at opusreader2/R/read_opus.R:213:3
 14.             └─base::lapply(dataset_list, calc_parameter_chunk_size) at opusreader2/R/parse_opus.R:148:5
 15.               └─opusreader2 (local) FUN(X[[i]], ...)
───────────────────────────────────────────────────────────────────────────────────────────────────────────────

══ Results ════════════════════════════════════════════════════════════════════════════════════════════════════
── Failed tests ───────────────────────────────────────────────────────────────────────────────────────────────
Error (test-read_opus.R:2:3): all test files are parsed without warning
Error in `UseMethod("calc_parameter_chunk_size")`: no applicable method for 'calc_parameter_chunk_size' applied to an object of class "parameter"
Backtrace:
     ▆
  1. ├─testthat::expect_no_warning(read_opus(dsn = "../../inst/extdata/test_data")) at test-read_opus.R:2:3
  2. │ └─testthat:::expect_no_(...)
  3. │   └─testthat:::quasi_capture(enquo(object), NULL, capture)
  4. │     ├─testthat (local) .capture(...)
  5. │     │ └─rlang::try_fetch(...)
  6. │     │   └─base::withCallingHandlers(...)
  7. │     └─rlang::eval_bare(quo_get_expr(.quo), quo_get_env(.quo))
  8. └─opusreader2::read_opus(dsn = "../../inst/extdata/test_data")
  9.   └─opusreader2:::opus_lapply(dsn, data_only) at opusreader2/R/read_opus.R:152:5
 10.     └─base::lapply(...) at opusreader2/R/read_opus.R:237:3
 11.       └─opusreader2 (local) FUN(X[[i]], ...)
 12.         └─opusreader2::read_opus_single(x, data_only) at opusreader2/R/read_opus.R:240:7
 13.           └─opusreader2:::parse_opus(raw, data_only) at opusreader2/R/read_opus.R:213:3
 14.             └─base::lapply(dataset_list, calc_parameter_chunk_size) at opusreader2/R/parse_opus.R:148:5
 15.               └─opusreader2 (local) FUN(X[[i]], ...)

[ FAIL 1 | WARN 0 | SKIP 0 | PASS 10 ]
> sessioninfo::session_info()
─ Session info ──────────────────────────────────────────────────────────────────────────────────────────────
 setting  value
 version  R version 4.3.2 (2023-10-31)
 os       macOS Sonoma 14.2.1
 system   aarch64, darwin20
 ui       RStudio
 language (EN)
 collate  en_US.UTF-8
 ctype    en_US.UTF-8
 tz       Europe/Zurich
 date     2023-12-23
 rstudio  2023.09.1+494 Desert Sunflower (desktop)
[... 43 lines cut ...]
   rprojroot     2.0.4   2023-11-05 [1] CRAN (R 4.3.1)
   rstudioapi    0.15.0  2023-07-07 [1] CRAN (R 4.3.0)
   sessioninfo   1.2.2   2021-12-06 [1] CRAN (R 4.3.0)
   shiny         1.8.0   2023-11-17 [1] CRAN (R 4.3.1)
   stringi       1.8.3   2023-12-11 [1] CRAN (R 4.3.1)
   stringr       1.5.1   2023-11-14 [1] CRAN (R 4.3.1)
   testthat    * 3.2.1   2023-12-02 [1] CRAN (R 4.3.1)
   tibble        3.2.1   2023-03-20 [1] CRAN (R 4.3.0)
   urlchecker    1.0.1   2021-11-30 [1] CRAN (R 4.3.0)
   usethis       2.2.2   2023-07-06 [1] CRAN (R 4.3.0)
   utf8          1.2.4   2023-10-22 [1] CRAN (R 4.3.1)
   vctrs         0.6.5   2023-12-01 [1] CRAN (R 4.3.1)
   waldo         0.5.2   2023-11-02 [1] CRAN (R 4.3.1)
   withr         2.5.2   2023-10-30 [1] CRAN (R 4.3.1)
   xfun          0.41    2023-11-01 [1] CRAN (R 4.3.1)
   xtable        1.8-4   2019-04-21 [1] CRAN (R 4.3.0)
   yaml          2.3.8   2023-12-11 [1] CRAN (R 4.3.1)

 [1] /Users/philipp/Library/R/arm64/4.3/library
 [2] /Library/Frameworks/R.framework/Versions/4.3-arm64/Resources/library

 P ── Loaded and on-disk path mismatch.

─────────────────────────────────────────────────────────────────────────────────────────────────────────────
```

philipp-baumann, 2023-12-23:
Hi @EstherDGoldstein-NOAA 

Thanks a lot for opening getting back to us with this issue. The issue what caused by a recent commit. Unfortunately, this was an annoying mistake on my side because of bad release practice (lesson learnt). We actually have the tests in place that would have picked what you report, since v0.6.1.

To fix, I did a patch release **v0.6.2**. I have some further explanations what was the root cause and how I hot-fixed it in the [**`NEWS` update**](https://github.com/spectral-cockpit/opusreader2/blob/main/NEWS.md) of today's version bumping. 

I ran the tests on the files I have, they pass. Would be great if you can reinstall the newest patch, and close this if it works.

Thanks for your understanding.

Cheers,
Philipp

EstherDGoldstein-NOAA, 2023-12-26:
Thank you so much for your quick response!

Esther

closed 2023-12-26 by EstherDGoldstein-NOAA