spectral-cockpit/opusreader2 #104: Error in strsplit(sample_name_long, split = ";") : non-character argument

dylanbeaudette, 2024-02-14:
Hi,

I'm having a strange issue with reading OPUS files and requesting only the data. This used to work, about a year ago:

```r
read_opus('e:/mir/MIR-training/HI-test/spectra/103283XS01.0', data_only = TRUE)
```

The error message looks like this:
```
Error in strsplit(sample_name_long, split = ";") : non-character argument
In addition: Warning messages:
1: In strptime(time_hour, format = "%Y/%m/%d %H:%M:%S", tz = etc_tz) :
  unknown timezone 'Etc/NA'
2: In as.POSIXct.POSIXlt(strptime(time_hour, format = "%Y/%m/%d %H:%M:%S",  :
  unknown timezone 'Etc/NA'
3: In as.POSIXlt.POSIXct(x) : unknown timezone 'Etc/NA'
```

Reading the file works as expected with `data_only = FALSE`.

esteveze, 2024-02-27:
I have the same issue:
data_list <- read_opus_single (dsn = "C:/Users/Documents/Data/POM characterization/FTIR/Data/DByDate/15_2_2024/L12_Y24.0")
Warning messages:
1: In strptime(time_hour, format = "%Y/%m/%d %H:%M:%S", tz = etc_tz) :
  unknown timezone 'Etc/GMT+1	'
2: In as.POSIXct.POSIXlt(strptime(time_hour, format = "%Y/%m/%d %H:%M:%S",  :
  unknown timezone 'Etc/GMT+1	'
3: In as.POSIXlt.POSIXct(x) : unknown timezone 'Etc/GMT+1	'

mtalluto, 2024-02-27:
For the warning in strptime, we tracked the problem down to unexpected whitespace in the timezone; in `get_meta_timestamp` our timezone is coming out "(GMT+1)\t" with a spare tab character. We solved it for the moment by updating a regular expression to strip whitespace from the end of the timezone (line 29 in extract_metadata.R):

```
tz <- gsub(pattern = "\\(|\\)|\\s+$", "", x = time_hour_tz[3L])
```

I'm reluctant to post this in a PR because I'm unsure where this tab character is coming from, and whether this is better handled somewhere in the parsing functions.

philipp-baumann, 2024-02-27:
> Hi,
>
> I'm having a strange issue with reading OPUS files and requesting only the data. This used to work, about a year ago:
>
> ```r
> read_opus('e:/mir/MIR-training/HI-test/spectra/103283XS01.0', [...]

Hi Dylan, thanks for bringing this issue up! I'll checkout a solution that fixes the *locale* setups we know, possibly this week.

philipp-baumann, 2024-02-27:
> For the warning in strptime, we tracked the problem down to unexpected whitespace in the timezone; in `get_meta_timestamp` our timezone is coming out "(GMT+1)\t" with a spare tab character. We [...]

Thanks for a first ad-hoc solution, i'll investigate a bit further and see whether we can use a fix that generalizes.

mtalluto, 2024-02-28:
Great, we can provide a sample file if needed.

philipp-baumann, 2024-02-28:
> Great, we can provide a sample file if needed.

please, very happy if you do so. baumann dash philipp at protonmail dot com or here.

dylanbeaudette, 2024-02-28:
Thanks everyone. I'll also send an example file!

philipp-baumann, 2024-02-29:
This should be fixed now :-) Would be great if you can briefly check out the PR and test with your example file, too. Also, would be good to have an extra test file. Good to not test CRAN size limit, but it can't hurt to solidify testing with reverse engineering ;-)

More infos what was done in PR.

@mtalluto regex parsing for extra tab/space was useful, and it provides stable to get rid of the warning.

Overall, we were missing to parse "sample" and "history" blocks when `data_only = TRUE`. Therefore, sample name and timezone infos from the OPUS files weren't included. It is good to do so I think.

philipp-baumann, 2024-02-29:
cc @ThomasKnecht , would be great if you can give a short review of the code (if time) :-)

philipp-baumann, 2024-02-29:
Thank you all for catching this bug 💯

philipp-baumann, 2024-02-29:
@mtalluto regarding the fix of the warning; the correct time info we could only find in the history. It is a messy file (Windows origin plus proprietary Bruker), so that's the best we can do to parse the text lines.

mtalluto, 2024-02-29:
Tested from my end, our file now loads with no warnings. I sent you the file via email, feel free to include with the tests.

closed 2024-03-12 by philipp-baumann

philipp-baumann, 2024-03-12:
> Thanks!

welcome, thanks also for the file. And if you have any suggestions or comments in the future, happy to discuss.

PR #105 (2024-03-12): fix metadata parsing `read_opus(dsn, data_only = TRUE)`
fixes #104 

- concerns sample name and correctly getting timezone info when `data_only = TRUE`. `data_only` is not strictly true, since we also get non-data blocks (sample and history). Still, I think it is a good default to do so and uniquely identify FT-IR measurements/replicates (sample name that was originally entered/correct time, since file operations of the OS change these attributes).

- added new tests to check for both warnings and errors, both for `read_opus(dsn, data_only = FALSE)` and `read_opus(dsn, data_only = TRUE)`, in a [... cut]